ntnu-arl/gbplanner_ros #4: libvolumetric_map_base.so: undefined reference to `cv::Mat::updateContinuityFlag()'

ArghyaChatterjee, 2020-07-23:
Hi, I followed your tutorial, made a gbplanner_ws workspace & tried to complie it (with and without octomapping). However, in both cases, the commands seems like having a mismatch in opencv version or something similar.

I am using Ubuntu 18.04 with ROS melodic running and an opencv version of 3.4.8. 
[build.make.003.log](https://github.com/unr-arl/gbplanner_ros/files/4967314/build.make.003.log)

```
arghya@arghya-Erazer-X7849-MD60379:~/gbplanner_ws$ catkin build -DCMAKE_BUILD_TYPE=Release gbplanner_ros
-------------------------------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/arghya/catkin_ws/devel:/home/arghya/ws_moveit/devel:/opt/ros/melodic
Workspace:                   /home/arghya/gbplanner_ws
-------------------------------------------------------------------------------------------------------
Build Space:        [exists] /home/arghya/gbplanner_ws/build
Devel Space:        [exists] /home/arghya/gbplanner_ws/devel
Install Space:      [unused] /home/arghya/gbplanner_ws/install
Log Space:          [exists] /home/arghya/gbplanner_ws/logs
Source Space:       [exists] /home/arghya/gbplanner_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release -DUSE_OCTOMAP=1
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------------------------------------------------
[build] Found '41' packages in 0.0 seconds.                                                                                                             
[build] Package table is up to date.                                                                                                                    
Starting  >>> catkin_simple                                                                                                                             
Starting  >>> planner_gazebo_sim                                                                                                                        
Starting  >>> planner_msgs                                                                                                                              
Finished  <<< planner_gazebo_sim                       [ 0.9 seconds ]                                                                                  
Finished  <<< catkin_simple                            [ 1.0 seconds ]                                                                                  
Starting  >>> eigen_catkin                                                                                                                              
Starting  >>> gflags_catkin                                                                                                                             
Starting  >>> kdtree                                                                                                                                    
Starting  >>> volumetric_msgs                                                                                                                           
Starting  >>> voxblox_msgs                                                                                                                              
Finished  <<< kdtree                                   [ 1.2 seconds ]                                                                                  
Finished  <<< gflags_catkin                            [ 1.2 seconds ]                                                                                  
Finished  <<< eigen_catkin                             [ 1.2 seconds ]                                                                                  
Starting  >>> glog_catkin                                                                                                                               
Finished  <<< planner_msgs                             [ 2.6 seconds ]                                                                                  
Starting  >>> planner_control_interface                                                                                                                 
Finished  <<< voxblox_msgs                             [ 1.8 seconds ]                                                                                  
Finished  <<< volumetric_msgs                          [ 1.8 seconds ]                                                                                  
Finished  <<< glog_catkin                              [ 1.0 seconds ]                                                                                  
Starting  >>> eigen_checks                                                                                                                              
Finished  <<< planner_control_interface                [ 1.5 seconds ]                                                                                  
Finished  <<< eigen_checks                             [ 1.0 seconds ]                                                                                  
Starting  >>> minkindr                                                                                                                                  
Finished  <<< minkindr                                 [ 0.7 seconds ]                                                                                  
Starting  >>> minkindr_conversions                                                                                                                      
Starting  >>> volumetric_map_base                                                                                                                       
Starting  >>> voxblox                                                                                                                                   
________________________________________________________________________________________________________________________________________________________
Warnings   << voxblox:cmake /home/arghya/gbplanner_ws/logs/voxblox/build.cmake.003.log                                                                  
INSTALL TARGETS - target voxblox_proto has PUBLIC_HEADER files but no PUBLIC_HEADER DESTINATION.
CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:418 (message):
  catkin_package() include dir '/home/arghya/gbplanner_ws/build/voxblox'
  should be placed in the devel space instead of the build space
[... 10 lines cut ...]
Finished  <<< volumetric_map_base                      [ 2.7 seconds ]                                                                                  
Starting  >>> octomap_world                                                                                                                             
Finished  <<< voxblox_rviz_plugin                      [ 1.9 seconds ]                                                                                  
Starting  >>> voxblox_ros                                                                                                                               
Finished  <<< voxblox_ros                              [ 3.5 seconds ]                                                                                  
________________________________________________________________________________________________________________________________________________________
Errors     << octomap_world:make /home/arghya/gbplanner_ws/logs/octomap_world/build.make.003.log                                                        
[... cut ...]
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                
[build]   Abandoned: 2 packages were abandoned.                                                                                                         
[build]   Failed:    1 packages failed.                                                                                                                 
[build] Runtime: 13.2 seconds total.
```

tungdanganh, 2020-08-13:
Hi, sorry for the late reply. I've prepared a branch for Melodic [here](https://github.com/unr-arl/gbplanner_ws/tree/melodic).
If you could please try the configuration setting in that branch first. Then if the problem still exists (which is very likely), I would suggest you use [this package](https://github.com/ethz-asl/opencv3_catkin) to install OpenCV3.2 in the workspace, which is the officially supported version in Melodic.

ArghyaChatterjee, 2020-08-14:
Hi @tungdanganh , thanks for responding. The issue seem to be resolved. However, it seems that there is a gazebo version mismatch:
```
arghya@arghya-Erazer-X7849-MD60379:~/gbplanner_ws$ catkin build
--------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/arghya/ws_moveit/devel:/opt/ros/melodic
Workspace:                   /home/arghya/gbplanner_ws
--------------------------------------------------------------------------
Build Space:        [exists] /home/arghya/gbplanner_ws/build
Devel Space:        [exists] /home/arghya/gbplanner_ws/devel
Install Space:      [unused] /home/arghya/gbplanner_ws/install
Log Space:          [exists] /home/arghya/gbplanner_ws/logs
Source Space:       [exists] /home/arghya/gbplanner_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------
[build] Found '41' packages in 0.0 seconds.                                                                     
[build] Package table is up to date.                                                                            
Starting  >>> catkin_simple                                                                                     
Starting  >>> lidar_description                                                                                 
Starting  >>> lidar_gazebo_plugins                                                                              
Starting  >>> mav_msgs                                                                                          
Starting  >>> mav_state_machine_msgs                                                                            
Starting  >>> mav_system_msgs                                                                                   
Starting  >>> planner_gazebo_sim                                                                                
Starting  >>> planner_msgs                                                                                      
Finished  <<< catkin_simple                            [ 0.1 seconds ]                                          
Finished  <<< mav_state_machine_msgs                   [ 0.3 seconds ]                                          
Finished  <<< lidar_description                        [ 0.2 seconds ]                                          
Finished  <<< mav_system_msgs                          [ 0.3 seconds ]                                          
Finished  <<< mav_msgs                                 [ 0.5 seconds ]                                          
Finished  <<< lidar_gazebo_plugins                     [ 0.2 seconds ]                                          
Starting  >>> rotors_comm                                                                                       
Finished  <<< planner_gazebo_sim                       [ 0.2 seconds ]                                          
Starting  >>> rotors_description                                                                                
Starting  >>> rotors_evaluation                                                                                 
Finished  <<< planner_msgs                             [ 0.9 seconds ]                                          
Finished  <<< rotors_comm                              [ 0.2 seconds ]                                          
Starting  >>> rqt_rotors                                                                                        
Starting  >>> eigen_catkin                                                                                      
Starting  >>> gflags_catkin                                                                                     
Starting  >>> kdtree                                                                                            
Starting  >>> volumetric_msgs                                                                                   
Starting  >>> voxblox_msgs                                                                                      
Finished  <<< rotors_description                       [ 0.1 seconds ]                                          
Starting  >>> yaml_cpp_catkin                                                                                   
Finished  <<< rotors_evaluation                        [ 0.1 seconds ]                                          
Starting  >>> mav_planning_msgs                                                                                 
Finished  <<< kdtree                                   [ 0.1 seconds ]                                          
Starting  >>> rotors_control                                                                                    
[... cut ...]
[build]   Warnings:  None.                                                                                      
[build]   Abandoned: 4 packages were abandoned.                                                                 
[build]   Failed:    2 packages failed.                                                                         
[build] Runtime: 43.7 seconds total.                                                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

tungdanganh, 2020-08-14:
Hi @ArghyaChatterjee,
Thanks for the feedback, just to make sure we are on the same page, have you tried with the [Melodic branch](https://github.com/unr-arl/gbplanner_ws/tree/melodic) in the gbplanner_ws?
In that branch, I added another version of RotorS ([Link](https://github.com/unr-arl/rotors_simulator/tree/dev/arl_planners_gazebo9) ) that should work with gazebo9 in Melodic.

ArghyaChatterjee, 2020-08-15:
Hi @tungdanganh ,
I actually tried to follow the steps inside the readme file in the melodic branch of the original repository which is : https://github.com/unr-arl/gbplanner_ws/tree/melodic
I can see that you haven't updated the readme file of that particular branch. Here is what I did:
```
sudo apt-get install libgoogle-glog-dev
sudo apt-get install ros-melodic-octomap*
git clone https://github.com/unr-arl/gbplanner_ws
cd gbplanner_ws
wstool init
wstool merge packages_https.rosinstall
wstool update
catkin config -DCMAKE_BUILD_TYPE=Release
catkin build
```
So, I got the gazebo version mismatch error which I pointed before. 

However, I tried to go with the steps mentioned in this website:
```
sudo apt-get install libgoogle-glog-dev
sudo apt-get install ros-melodic-octomap*
git clone https://github.com/unr-arl/gbplanner_ws
cd gbplanner_ws
git checkout melodic-devel
wstool init
wstool merge packages_https.rosinstall
wstool update
catkin config -DCMAKE_BUILD_TYPE=Release
catkin build
```
So when I initiate the command ```git checkout melodic-devel``` it shows another error:
```
arghya@arghya-Erazer-X7849-MD60379:~/gbplanner_ws$ git checkout melodic-devel
error: pathspec 'melodic-devel' did not match any file(s) known to git.
```

tungdanganh, 2020-08-16:
The branch is named ```melodic```
Could you please try again with ```git checkout melodic```?

ArghyaChatterjee, 2020-08-16:
I tried with ```git checkout melodic```  and it did work but yielding the same issue ```libvolumetric_map_base.so: undefined reference to `cv::Mat::updateContinuityFlag()```. Although 34/37 packages succeeded after initiating catkin build. Same is the case with 'mbplanner' package when I try to run on ROS melodic.

However, I have tried to build the package on a Ubuntu 16.04 machine with ROS kinetic running and everything went smooth. I was having the default version of opencv 3.2 which was automatically installed during ROS installation. I had made a series of video earlier on youtube on [gbplanner](https://www.youtube.com/watch?v=2jMq3GOhbbo&list=PLVy6YSUUzzp1SDggmxFhSsXsifBs9w6Q0).  

So, this may be an issue regarding to opencv version mismatch which is in my case 3.4.8.

tungdanganh, 2020-08-16:
Thanks @ArghyaChatterjee.

closed 2020-08-16 by tungdanganh