ntnu-arl/gbplanner_ros #5: Using RGB-D Camera

engcang, 2020-12-08:
Hi. Thanks for sharing this great repo.

I think this repo is really similar with [MBPlanner](https://github.com/unr-arl/mbplanner_ros/pull/5) from your lab.

So, the same issue is happening here.
Using RGB-D Camera with "kCamera" `SensorParams` in config files,
It continuously returns "No positive gain was found"

[ WARN] [1607462735.649574133, 16.540000000]: [PLANNER_ERROR] No positive gain was found.
[ WARN] [1607462735.649621397, 16.540000000]: Planner returned an empty path
[ INFO] [1607462736.207201124, 17.040000000]: PlannerControlInterface: Running Planner (kAuto)
[ INFO] [1607462736.218772441, 17.052000000]: Planning iteration 3
[ INFO] [1607462736.267533763, 17.100000000]: Set bound mode: kExtendedBound
[ INFO] [1607462736.268022768, 17.100000000]: Current box contains Occupied voxels.

I think this repo also has to be fixe as [this pull request of MBPlanner](https://github.com/unr-arl/mbplanner_ros/pull/5).
As I made a same issue there, and got fixed from the first author of the paper [here](https://github.com/unr-arl/mbplanner_ros/issues/4)

Thank you in advance.

engcang, 2020-12-08:
Ohhh Sorry I looked into the code carefully and I found that [this pull request of MBP](https://github.com/unr-arl/mbplanner_ros/pull/5) was solved thanks to referring this repo.

And by setting bigger `Local` and `LocalSearch` made it work with `kCamera`.

Thank you!

closed 2020-12-08 by engcang

ShreyanshDarshan, 2021-03-18:
@engcang 
Can you share your `Local` and `LocalSearch` values that made it work on `kCamera`?
Also, which RGBD camera did you use on your vehicle?

engcang, 2021-03-19:
@ShreyanshDarshan  
Hi. I used Rotors simulator as this repo originally adopted it, so I just changed the specification of the camera in `component_snippets.xacro` rather than using other camera.
For the `Local` and `LocalSearch`, I used the twice of `max_range` for X and Y axes, and 1.5, 3.0 for Z axis values each.

But I felt that the bigger values make it work better (e.g., [50, 50, 1.5]), and also resulted in the longer computation time.