ntnu-arl/gbplanner_ros #7: pci_mav package crashes on using gcc-8 and g++-8

mihirk284, 2021-02-10:
Using gcc-8 and g++-8 causes the pci_mav package to crash after executing 1 planning iteration. 

Using gcc-7 and g++-7 does not cause this error to show up.

```
[ WARN] [1612929626.567301264, 26.350000000]: PCI: Ready to trigger the planner.
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[pci_mav_ros_node-16] process has died [pid 21673, exit code -6, cmd /home/arl/gbplanner_ws/devel/lib/pci_mav/pci_mav_ros_node command/trajectory:=m100/command/trajectory planner_server:=gbplanner planner_homing_server:=gbplanner/homing odometry:=m100/ground_truth/odometry_throttled __name:=pci_mav_ros_node __log:=/home/arl/.ros/log/6f136fca-6b54-11eb-8bb6-7085c2cb6201/pci_mav_ros_node-16.log].
log file: /home/arl/.ros/log/6f136fca-6b54-11eb-8bb6-7085c2cb6201/pci_mav_ros_node-16*.log
```

MihirDharmadhikari, 2021-02-10:
Thank you for pointing this out! We are currently investigating on the cause of this error.

wwtinwhu, 2021-02-19:
> Thank you for pointing this out! We are currently investigating on the cause of this error.

I also met this problem.  I will try to debug the pci_mav_ros_node and find the code crashes. Hope this can help.

wwtinwhu, 2021-03-04:
> Thank you for pointing this out! We are currently investigating on the cause of this error.

@MihirDharmadhikari Hi, I try to debug the pci_mav_ros_node in CLion. But it is too hard to debug the node in Clion with some parameters need to be setted in launch file. Do you any advice? How do you debug code in ROS? Thank you.

MihirDharmadhikari, 2021-03-04:
Hi @wwtinwhu ,
Thank you for your interest in debugging the package. I have used the gdb to debug the ROS node but it only provides the stack trace and will give you the function which caused the crash. I am currently looking into the VSCode ROS plugin debugger that allows tracking of variables as well.

MihirDharmadhikari, 2021-04-04:
This issue was resolved through the Pull Request ntnu-arl/pci_mav#1.
The problem was that the [executePath](https://github.com/ntnu-arl/pci_mav/blob/3035d0caeb0659695603f7303af04a075e2679f2/src/pci_mav.cpp#L271) function was not returning anything upon successful execution but the return type was bool. This has now been fixed and stricter cmake flags have been added to the CMakeLists.txt to catch such issues during build time.

closed 2021-04-04 by MihirDharmadhikari