ntnu-arl/gbplanner_ros #8: Exploration in Outdoor Environments

derektan95, 2021-02-19:
Hi, 

I've studied both your papers and your codebase for awhile now. Amazing work here! I've 2 questions: 

(1) Performance in outdoor environments
From my understanding, the algorithm works on choosing paths with highest exploration gain, which is a function of volumetric gain, while minimizing heading changes and path distance.  

I would like to ask if your algorithm has been tested in outdoor environments? For instance, an area with many buildings (e.g. University campus, residential complex), or nature areas (e.g. forest)? How efficient is the path taken by the robot in mapping such an environment with many possible branching paths? 

(2) Multi-Agent Exploration
Can your algorithm be used by multiple robots to perform concurrent exploration? Perhaps the map can be shared in real time between the robots for faster exploration completion, and the different robots can take charge of exploring different areas? 

Thanks alot, 
Derek

MihirDharmadhikari, 2021-02-19:
Hi @derektan95 ,

Thank you for your interest in our work!

1. As the planner works on maximizing the unknown volume perceived, it will work well in outdoor environments as well.  However, as this work was tailored towards and motivated by subterranean environments, we haven't done extensive testing in outdoor environments as we have done in indoor and subterranean environments. So there are no performance benchmarks available in outdoor environments as what we have for subterranean environments.
One challenge that I see is longer mapping times due to open space. You may want to reduce the `max_ray_length_m` parameter in the [voxblox_sim_config.yaml](https://github.com/ntnu-arl/gbplanner_ros/blob/master/gbplanner/config/m100/voxblox_sim_config.yaml) if you are using voxblox or `sensor_range_max` in [octomap_sim_config.yaml](https://github.com/ntnu-arl/gbplanner_ros/blob/master/gbplanner/config/m100/octomap_sim_config.yaml) if using octomap to speed up the mapping.
2. Regarding your second question, the planner is not currently aware of the exploration of other robots. Hence, it won't work as it is for multi-robot exploration. Sharing an entire volumetric map between the robots will be expensive too. We are currently investigating on multi-robot exploration planning methods for subterranean environments and will release our work once complete.

I hope this information helps. Let me know if you have any other questions.

Thanks,
Mihir

derektan95, 2021-03-03:
Hi Mihir, 

Thanks for the information provided, really helpful! I will try this out. Good luck with your endeavors!

-Derek

derektan95, 2021-09-11:
Hi @MihirDharmadhikari, 

Thought I would like to ask this since we were on the topic of multi-robot exploration. Was excited when I saw that your team released the multi-robot version of the gbplanner, which integrates with the COHORT central planner. I wonder if this work will be released anytime soon? Would love to try it out. 

P.s. glad that you are using PCA to derive the bounding box for RRT to work in! 

Regards,
Derek

MihirDharmadhikari, 2021-11-02:
Hi @derektan95 ,

Thank you for your interest in our work on COHORT. We do plan to release this eventually but we don't have a specific timeline yet.

Best,
Mihir

derektan95, 2021-11-27:
Hi @MihirDharmadhikari, 

Thanks alot! I'll stay tuned.

closed 2021-11-27 by derektan95