ntnu-arl/gbplanner_ros #13: Input pointcloud queue getting too long!

zyhphd, 2021-12-07:
When i try to run a demo, i met the error as follows, 

QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ERROR] [1638882589.949854752, 60.176000000]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.
[ERROR] [1638882627.590240752, 120.192000000]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.
[ERROR] [1638882664.958607277, 180.300000000]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.

Who met the same error before? Thanks a lot.

MihirDharmadhikari, 2021-12-07:
Hi @zyhphd,

When you run the demo, do the buttons in the UI work? If you are able to interact with the planner using the UI (as described in the [wiki](https://github.com/ntnu-arl/gbplanner_ros/wiki/Demo)) you can safely ignore the `Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long` error. This error is raised by voxblox. If voxblox is not building the map at all, then there is an issue.
Let me know if it works.

Best,
Mihir

zyhphd, 2021-12-08:
> Hi @zyhphd,
>
> When you run the demo, do the buttons in the UI work? If you are able to interact with the planner using the UI (as described in the [...]

zyhphd, 2021-12-08:
I click the button(start planner), but nothing work , the planner failed and reported the error as follows,  i miss some important file or config?  

Warning [parser.cc:950] XML Element[pbr], child of element[material] not defined in SDF. Ignoring[pbr]. You may have an incorrect SDF file, or an sdformat version that doesn't support this element.
[ INFO] [1638928583.864289327]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1638928583.877666131]: Physics dynamic reconfigure ready.
[INFO] [1638928583.979126, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1638928614.979755, 875.948000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name rmf_obelix
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ERROR] [1638928614.980698, 875.948000]: Spawn service failed. Exiting.
[rmf_obelix/spawn_rmf_obelix-12] process has died [pid 24020, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -param robot_description -urdf -x 0.0 -y 0.0 -z 0.5 -model rmf_obelix __name:=spawn_rmf_obelix __log:=/home/zyh/.ros/log/0a95bffe-57ca-11ec-b518-b42e99addfdd/rmf_obelix-spawn_rmf_obelix-12.log].
log file: /home/zyh/.ros/log/0a95bffe-57ca-11ec-b518-b42e99addfdd/rmf_obelix-spawn_rmf_obelix-12*.log
[ INFO] [1638928617.262658769, 875.948000000]: Lidar laser plugin ready, 32 lasers
[ INFO] [1638928617.748424087, 876.680000000]: LeePositionController got first odometry message.
[ERROR] [1638928627.469152452, 894.736000000]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.
[ INFO] [1638928648.047199540, 933.156000000]: Setting goal: Frame:world, Position(10.710, 2.073, 0.000), Orientation(0.000, 0.000, 0.000, 1.000) = Angle: 0.000

[ INFO] [1638928657.147227899, 950.172000000]: Planning iteration 0
[ERROR] [1638928659.752033433, 954.764000000]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.
gbplanner_node: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:365: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
*** Aborted at 1638928672 (unix time) try "date -d @1638928672" if you are using GNU date ***
PC: @     0x7fa620048fb7 gsignal
*** SIGABRT (@0x3e800005de1) received by PID 24033 (TID 0x7fa6230f3280) from PID 24033; stack trace: ***
    @     0x7fa620049040 (unknown)
    @     0x7fa620048fb7 gsignal
    @     0x7fa62004a921 abort
    @     0x7fa62003a48a (unknown)
    @     0x7fa62003a502 __assert_fail
    @     0x7fa622ba7fa5 Eigen::DenseCoeffsBase<>::operator()()
    @     0x7fa622ba65fe _ZN17MapManagerVoxbloxIN7voxblox10TsdfServerENS0_9TsdfVoxelEE13getScanStatusERN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEERSt6vectorIS6_SaIS6_EERSt5tupleIJiiiEERS8_ISt4pairIS6_N10MapManager11VoxelStatusEESaISI_EER16SensorParamsBase
    @     0x7fa620e8c0c6 explorer::Rrg::computeVolumetricGainRayModel()
    @     0x7fa620e8a837 explorer::Rrg::computeExplorationGain()
    @     0x7fa620e7dad7 explorer::Rrg::evaluateGraph()
    @     0x7fa620e24619 explorer::Gbplanner::plannerServiceCallback()
    @     0x7fa620e4fde2 boost::_mfi::mf2<>::operator()()
    @     0x7fa620e487f6 boost::_bi::list3<>::operator()<>()
    @     0x7fa620e429d9 boost::_bi::bind_t<>::operator()<>()
    @     0x7fa620e3ce70 boost::detail::function::function_obj_invoker2<>::invoke()
    @     0x7fa620e698c7 boost::function2<>::operator()()
    @     0x7fa620e6695b ros::ServiceSpec<>::call()
    @     0x7fa620e5fc7d ros::ServiceCallbackHelperT<>::call()
    @     0x7fa621567974 ros::ServiceCallback::call()
    @     0x7fa6215bb829 ros::CallbackQueue::callOneCB()
    @     0x7fa6215bd5cb ros::CallbackQueue::callAvailable()
    @     0x7fa621614cf9 ros::SingleThreadedSpinner::spin()
    @     0x7fa6215fd62b ros::spin()
    @     0x563ee9d978d8 (unknown)
    @     0x7fa62002bbf7 __libc_start_main
    @     0x563ee9d975ba (unknown)
[ERROR] [1638928672.715952425, 981.480000000]: Planner service failed
[gbplanner_node-15] process has died [pid 24033, exit code -6, cmd /home/zyh/zyh/code/gbplanner/devel/lib/gbplanner/gbplanner_node odometry:=rmf_obelix/ground_truth/odometry_throttled /pointcloud:=/rmf_obelix/velodyne_points __name:=gbplanner_node __log:=/home/zyh/.ros/log/0a95bffe-57ca-11ec-b518-b42e99addfdd/gbplanner_node-15.log].
log file: /home/zyh/.ros/log/0a95bffe-57ca-11ec-b518-b42e99addfdd/gbplanner_node-15*.log
[ INFO] [1638928672.844882790, 981.996000000]: Planning iteration 0
[ERROR] [1638928672.844939585, 981.996000000]: Planner service failed
[ INFO] [1638928672.974171212, 982.508000000]: Planning iteration 0
[ERROR] [1638928672.974217385, 982.508000000]: Planner service failed
[ INFO] [1638928673.103324438, 983.024000000]: Planning iteration 0
[ERROR] [1638928673.103371541, 983.024000000]: Planner service failed
[... cut ...] ning iteration 0
[ERROR] [1638928673.231575202, 983.536000000]: Planner service failed
[ INFO] [1638928673.359649981, 984.048000000]: Planning iteration 0
[ERROR] [1638928673.359704760, 984.048000000]: Planner service failed
[ INFO] [1638928673.488844883, 984.564000000]: Planning iteration 0
[ERROR] [1638928673.488885598, 984.564000000]: Planner service failed
[ INFO] [1638928673.617040537, 985.076000000]: Planning iteration 0
[ERROR] [1638928673.617084928, 985.076000000]: Planner service failed

MihirDharmadhikari, 2021-12-08:
Hello,

It seems that the node is crashing during the volumetric gain evaluation step. Can you check two things:
- Can you visualize the occupied voxels in rviz to make sure that it is building a map? It will be under the `voxblox` group in the rviz config provided with the demo. The topic to visualize is `/gbplanner_node/occupied_nodes`.
- Can you change the planner verbosity to `DEBUG` as explained [here](https://github.com/ntnu-arl/gbplanner_ros/wiki/Parameters#verbosity), run the planner again, and send me the terminal log? You will have to change [this](https://github.com/ntnu-arl/gbplanner_ros/blob/d9ecc1f6ab597a6beb183ef058280ad207dc5bf9/planner_common/include/planner_common/params.h#L24) parameter and compile again.

Let me know how it goes.

Best,
Mihir

zyhphd, 2021-12-08:
Thanks a lot for your reply. I check the visualization of the occupied voxels, it works well. Then, i change the planner verbosity to DEBUG and the terminal log as follows:

[ INFO] [1638941521.339947665, 876.404000000]: LeePositionController got first odometry message.
[ WARN] [1638941521.391149221, 876.496000000]: Planner is waiting for odometry
[ WARN] [1638941521.560278677, 876.852000000]: Planner is waiting for odometry
[ WARN] [1638941521.810180202, 877.316000000]: Received the first odometry, reset the map
[ERROR] [1638941531.876572623, 896.012000000]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.
[ WARN] [1638941555.902195485, 940.456000000]: Planner Trigger Mode set to kManual.
[ERROR] [1638941564.604267060, 956.184000000]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.
[ WARN] [1638941576.177352949, 976.184000000]: Planner Trigger Mode set to kAuto.
[ INFO] [1638941576.192390747, 976.196000000]: Planning iteration 0
[ INFO] [1638941582.462378040, 987.116000000]: Set bound mode: kExtendedBound
[ INFO] [1638941587.554860561, 996.972000000]: Formed a graph with [456] vertices and [9001] edges with [576] loops
[ INFO] [1638941587.566944896, 997.020000000]: [Vis]: Num hanging verts: 0
gbplanner_node: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:365: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
*** Aborted at 1638941587 (unix time) try "date -d @1638941587" if you are using GNU date ***
PC: @     0x7fa38a89efb7 gsignal
*** SIGABRT (@0x3e80000749b) received by PID 29851 (TID 0x7fa38d949280) from PID 29851; stack trace: ***
    @     0x7fa38a89f040 (unknown)
    @     0x7fa38a89efb7 gsignal
    @     0x7fa38a8a0921 abort
    @     0x7fa38a89048a (unknown)
    @     0x7fa38a890502 __assert_fail
    @     0x7fa38d3fe04b Eigen::DenseCoeffsBase<>::operator()()
    @     0x7fa38d3fc00c _ZN17MapManagerVoxbloxIN7voxblox10TsdfServerENS0_9TsdfVoxelEE13getScanStatusERN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEERSt6vectorIS6_SaIS6_EERSt5tupleIJiiiEERS8_ISt4pairIS6_N10MapManager11VoxelStatusEESaISI_EER16SensorParamsBase
    @     0x7fa38b6e2272 explorer::Rrg::computeVolumetricGainRayModel()
    @     0x7fa38b6e09dd explorer::Rrg::computeExplorationGain()
    @     0x7fa38b6d3b9f explorer::Rrg::evaluateGraph()
    @     0x7fa38b67a64f explorer::Gbplanner::plannerServiceCallback()
    @     0x7fa38b6a5e5c boost::_mfi::mf2<>::operator()()
    @     0x7fa38b69e870 boost::_bi::list3<>::operator()<>()
    @     0x7fa38b698a53 boost::_bi::bind_t<>::operator()<>()
    @     0x7fa38b692eea boost::detail::function::function_obj_invoker2<>::invoke()
    @     0x7fa38b6bf941 boost::function2<>::operator()()
    @     0x7fa38b6bc9d5 ros::ServiceSpec<>::call()
    @     0x7fa38b6b5cf7 ros::ServiceCallbackHelperT<>::call()
    @     0x7fa38bdbd974 ros::ServiceCallback::call()
    @     0x7fa38be11829 ros::CallbackQueue::callOneCB()
    @     0x7fa38be135cb ros::CallbackQueue::callAvailable()
    @     0x7fa38be6acf9 ros::SingleThreadedSpinner::spin()
    @     0x7fa38be5362b ros::spin()
    @     0x56343d3408d8 (unknown)
    @     0x7fa38a881bf7 __libc_start_main
    @     0x56343d3405ba (unknown)
[ERROR] [1638941587.673091363, 997.212000000]: Planner service failed
[gbplanner_node-15] process has died [pid 29851, exit code -6, cmd /home/zyh/zyh/code/gbplanner/devel/lib/gbplanner/gbplanner_node odometry:=rmf_obelix/ground_truth/odometry_throttled /pointcloud:=/rmf_obelix/velodyne_points __name:=gbplanner_node __log:=/home/zyh/.ros/log/158cb390-57e8-11ec-b518-b42e99addfdd/gbplanner_node-15.log].
log file: /home/zyh/.ros/log/158cb390-57e8-11ec-b518-b42e99addfdd/gbplanner_node-15*.log
[ INFO] [1638941587.802022352, 997.728000000]: Planning iteration 0
[ERROR] [1638941587.802060469, 997.728000000]: Planner service failed

Best,
zyh

zyhphd, 2021-12-08:
Hi Mihir, after adding lots of std::cout, i finally find that the error was caused by the function in voxblox_alt_impl.cpp: 
if (x_increment_dist <= voxel_size && x < width)
            starting_points(x + 1, y) = std::min(ray_norm, step);
          std::cout<<"start getScanStatus 5.3.1"<<std::endl;
          if (y_increment_dist <= voxel_size && y > 0)
            starting_points(x, y - 1) = std::min(ray_norm, step);
          std::cout<<"start getScanStatus 5.3.2"<<std::endl;
          if (diag_increment_dist <= voxel_size && x < width && y > 0)
            starting_points(x + 1, y - 1) = std::min(ray_norm, step);
It seems that x+1 exceed the limit? x < width need to be changed to x + 1 < width ?

zyhphd, 2021-12-08:
Hi Mihir, after adding lots of std::cout, i finally find that the error may be caused by the function getScanStatus in voxblox_alt_impl.cpp, where starting_points(x , y) should be starting_points(y , x)? Because you define starting_points(height, width), but the loop is 
for (int y = height - 1; y >= 0; --y) {
    for (int x = 0; x < width; ++x) {

MihirDharmadhikari, 2021-12-08:
Hello,

Thank you for pointing this out. I will look into this and get back to you.

Best,
Mihir

zyhphd, 2021-12-15:
Hi Mihir,
 Is there an error?

ArghyaChatterjee, 2021-12-26:
Hi @MihirDharmadhikari , I am facing the same issue. With the command `roslaunch gbplanner rmf_sim.launch`, I am having the same issue as @zyhphd . I can see the sensor reading (VLP) when setting the global options `fixed frame` to `rmf_obelix/base_link` but just after launch, I can't see any occupied voxel / voxblox mesh under voxblox panel ( followed initialization ---> start planner in the rviz gui but nothing happens). When enabled the gazebo, I saw the robot model kinds of very slightly vibrating when commanded initialization and start planner.
[image]

[image]

[image]
[image]

Kucukcollu, 2021-12-28:
I have same issue with @ArghyaChatterjee . While mobile robot working well, the drone does not working.

zyhphd, 2021-12-29:
> Hi @MihirDharmadhikari , I am facing the same issue. With the command `roslaunch gbplanner rmf_sim.launch`, I am having the same issue as @zyhphd . I can see the sensor reading (VLP) when setting [...]

Maybe you can try to download and build the code one more time, i install and build the code by the step of Installation introduction again, and everything works well.

Kucukcollu, 2021-12-31:
Hi guys,
I am currently using Ubuntu 18.04 with ROS Melodic. And I realized that actual error is SDF version difference between my systems SDF version and model.sdf which is downloaded in the installation step. So I removed sdf from my system and reinstall the SDF. You can also try this method to be sure.

To completely remove SDF from your system run the following commands(I run all commands to be sure :smile:  ):

sudo apt-get remove  sdf
sudo apt-get remove --auto-remove sdf
sudo apt-get purge sdf
sudo apt-get purge --auto-remove sdf

Reinstall with these commands:

sudo apt-get update
sudo apt-get install sdf

zyhphd, 2021-12-31:
> Hi guys, I am currently using Ubuntu 18.04 with ROS Melodic. And I realized that actual error is SDF version difference between my systems SDF version and model.sdf which is downloaded in the installation step. So I removed sdf from my system and reinstall the SDF. You can also try this method to be sure.
> 
> To completely remove SDF from your system run the following commands(I run all commands to be sure 😄 ):
> 
> sudo apt-get remove sdf sudo apt-get remove --auto-remove sdf sudo apt-get purge sdf sudo apt-get purge --auto-remove sdf
> 
> Reinstall with these commands:
> 
> sudo apt-get update sudo apt-get install sdf

Thanks for that. And i have solved the problem.

MihirDharmadhikari, 2022-01-01:
Hello everyone,

Sorry for the delay from my side. It seems that most of you were able to solve the issue and I thank you all for the solutions. However, since it was faced by many, I will add a compilation of the above issue and fixes (like reinstalling sdf as suggested by @Kucukcollu ) in the wiki as possible debugging steps. 

Best,
Mihir

closed 2022-01-02 by zyhphd