ntnu-arl/gbplanner_ros #14: Integration with other SLAM Algorithms

derektan95, 2021-12-31:
Hi @MihirDharmadhikari, 

I wonder if this graph-based planner can work with other types of SLAM algorithms such as [RTABMap](https://github.com/introlab/rtabmap_ros)? It has the flexibility to allow us to map the scene with RGBD input, while localizing using 3D LiDAR input. I don't think there's such an option for Vox Graph. Thanks.

P.s. I recently had the chance to watch your thesis presentation, congratulations!

Regards, 
Derek

MihirDharmadhikari, 2022-01-01:
Hello @derektan95 ,

Currently GBPlanner2 works independent of the localization method. So RTABMap can be used for localization. However, for volumetric mappingGBPlanner2 only supports Voxblox due to the optimizations done in the planner for the same. As long as the correct transform from the sensor frame to the world frame exists in the tf tree, the Voxblox server will build the correct volumetric map irrespective of the localization method used. As you correctly pointed out, GBPlanner1 supports both Octomap and Voxblox, but GBPlanner2 only supports Voxblox. 

In future we do plan to integrate GBPlanner with the existing ROS frameworks like the ROS Navigation Stack to address a broader user community.

Similar to GBPlanner, COHORT also works independent of the robot's onboard localization solution hence RTABMap can be used onboard the robots for localization.

I hope this helps.

Best,
Mihir

derektan95, 2022-01-01:
Hi @MihirDharmadhikari, 

Great to hear from you. Just a couple of follow-up questions: 

**1) Is it efficient to run both VoxGraph and Rtabmap at the same time?** 
For RTABMap, I have configured it to run Iterative Closest Point on consecutive 3d lidar scans to derive accurate SLAM pose (i.e. sensor frame -> world frame). Similarly, it seems to me that VoxGraph operates by converting pointcloud input into Sign Distance Function submaps, which are geometrically aligned to each other to derive the robot's SLAM pose. 

_Based on your suggestion, I could use RTABMap to derive the sensor -> world frame, while generating the voxblox volumetric map using the Voxblox server (I assume voxblox params are specified here in [this launch file](https://github.com/ntnu-arl/gbplanner_ros/blob/08a8c74f14e49572fa2c10cadf3c0b19ef2b6476/gbplanner/launch/smb/smb_gazebo.launch#L49))._ But won't running core gbplanner launch file (which pressumably runs VoxGraph) perform redundant work that RTABMap is already doing? In other words, won't it be wasted effort to generate SDF submaps and perform geometric alignment to derive the robot's SLAM pose when RTABMap is already doing it? Are there parameters to set such that VoxGraph will not perform unnecessary steps when accurate sensor -> world frame transforms are already derived from other SLAM algorithms (like RTABMap)?

**2) In your real physical setup in the SubT challenge, are the robots all connected 1 single ROSCore on the central computer?** 
Or do you have roscores running on each robot (central computer included), and have them connected using [packages](http://wiki.ros.org/multimaster_fkie) like these?  It seems to be that some ROS nodes gets killed when it can't detect the ROSMaster, which is against the resilience that this repository is aiming for. 

Thanks as always, 
Derek

MihirDharmadhikari, 2022-01-02:
Hi Derek,

GBPlanner does **not** use VoxGraph. It uses Voxblox, which is used only for volumetric mapping and takes the sensor -> world transform provided externally. Hence, it will not do the redundant calculation of deriving this transform. The launchs file provided in the demos only launch Voxblox server, VoxGraph is not used anywhere.

Hope this helps.

Best,
Mihir

derektan95, 2022-01-02:
Hi @MihirDharmadhikari, 

Thanks for the prompt response. I am getting a much better understanding of this repository now. I understand now that GBPlanner2 only uses VoxBlox for volumetric mapping at the sensor -> world tf provided by any external source. 

**Am I then right to say that VoxGraph is used only in COHORT?**
I am getting this impression because [this COHORT explanation video](https://youtu.be/8IPdxcbhp5o?t=777) specified the use of VoxGraph as a centralized mapping framework to align and fuse VoxBlox submaps from each robot. This merged map is used to identify and assign exploration volumes for robots to explore in. [Your paper](https://arxiv.org/pdf/2111.06482.pdf) suggest likewise.

Thanks as always, 
Derek

MihirDharmadhikari, 2022-01-03:
Hi Derek,

Yes, Voxgraph is only used by COHORT to align and merge the submaps from all the robots on a centralized ground station. Whereas GBPlanner onboard each robot only uses Voxblox as the volumetric mapping framework.

Best,
Mihir

derektan95, 2022-01-17:
Hi @MihirDharmadhikari, 

That's great. I have a much better understanding of this project now, after having read the VoxBlox and VoxGraph papers too. Looking forward to trying out COHORT when it's out. Good luck with your academic endeavours!

Regards, 
Derek

closed 2022-01-17 by derektan95