ntnu-arl/gbplanner_ros #16: What is the GBPlanner Control?

zzhmx, 2022-02-04:
Hello, I am looking at your open source project: NTNU Autonomous Robots Lab. I followed the above tutorial to run the Aerial Robot Demo. When I input: "roslaunch gbplanner rmf_sim.launch", the rviz interface is displayed, but no The point cloud is displayed,and the error message is :" Error:Could not find library corresponding to plugin gbplanner_ui/GBPlanner Control.Make sure the plugin description XML file has the correct name of the library and that the library actually exists. ", but I can't find the GBPlanner Control, how can I solve it. thanks

MihirDharmadhikari, 2022-02-04:
Hello,

It looks like it could not find the gbplanner_ui package. Can you check if that package is built properly?

Best,
Mihir

zzhmx, 2022-02-05:
hello,
I recompiled once, it took a long time this time, but no error was reported, it showed that there were 3 ignored and 4 warnings, but after I ran it, the original error was still there.
thanks

zzhmx, 2022-02-08:
Hello, I recompiled and found that some files were missing. There was no bug after installation. thank you

closed 2022-02-08 by zzhmx