ntnu-arl/gbplanner_ros #20: Wrong subscript order in voxblox_alt_impl.cpp

SunZezhou, 2022-03-11:
Hello, 
    Thank you very much for your work. When I was reading the code I noticed that the order of the subscript of `starting_points` in [gbplanner_ros/planner_common/src/voxblox/voxblox_alt_impl.cpp](https://github.com/ntnu-arl/gbplanner_ros/tree/gbplanner2/planner_common/src/voxblox/voxblox_alt_impl.cpp) seems to be written in reverse order. 

The `starting_points(height, width)` is declared on line 569. 
But the subscript is `starting_points(width, height)`, as shown below. 
It seems that the order of all `starting_points` in the file may need to be swapped. 

[image]

In addition, I think the range of `x` should be [0, width - 1), instead of [0, width) (line 606). 
Otherwise `starting_points(x + 1, y)` will also encounter the problem of out-of-bounds, such as line 693. 
Please have a check. 

Best, 
Zezhou

MihirDharmadhikari, 2022-03-24:
Hi Zezhou,

I will look into this and get back to you.

Best,
Mihir

Ray005, 2024-12-02:
Could I ask what command you used to build the project? Did you use ```catkin_make``` or ```catkin build```?

I encountered the same issue when using ```catkin_make```, but it works fine with ```catkin build```.

MihirDharmadhikari, 2024-12-02:
Hi @Ray005,

We use `catkin build` for the our project. You can find the exact commands in the readme or in the wiki of the repository.

Best,
Mihir