ntnu-arl/gbplanner_ros #21: command interface using px4

DavidePatria, 2022-06-09:
I am trying to use this planner with px4 as I would like to simulate a drone I will later build.
I was looking for the actual commands sent to the drone, which are usually passed as `mavros_msgs/AttitudeTarget` or as `trajectory_msgs/MultiDOFJointTrajectoryPoint`. 
Analyzing the topic `/rmf_obelix/command/trajectory` I found that there a message of the latter type is published in it, but at a very slow rate and probably not for the purpose at which such messages are normally published, so I have a doubt about it. Is it used to keep a rough track of 

`/rmf_obelix/command/motor_speed` instead publishes a message of type `mav_msgs/Actuators` direclty to gazebo, which I understand is the used way to move the drone.

now, is there a way to get the planner to output one this type of messages among the two mentioned above?

MihirDharmadhikari, 2022-06-27:
Hello @DavidePatria ,

Sorry for the late reply. The final path is published by the Planner Control Interface (PCI) which publishes it as a `trajectory_msgs/MultiDOFJointTrajectoryPoint`. This is published only when a new path is calculated. Since the PX4 might need a continuous stream of input the way we handle this is by using a separate controller that will use the trajectory published by PCI and convert it to attitude commands. If you want to use the position controller of the PX4 stack you can have a path tracker that publishes waypoints instead of attitude commands. 
In conclusion, PCI only publishes the path when a new path is calculated and a separate controller is need to follow this path.

Let me know if this helps.

Best,
Mihir

DavidePatria, 2022-07-04:
Hi, thank you for your response.
I was able to identify and connect the right topics to the components I an truing to use, but the published waypoints seem too few to trigger px4 and the drone doesn't follow the path properly.  I am using a controller that takes the correct path message

MihirDharmadhikari, 2022-07-05:
What commands does your controller provide to the px4? If you are sending position waypoints to px4, then your controller will have to send one waypoint at a time and send the next waypoint only after the robot has reached the first one.

DavidePatria, 2022-07-05:
I am using [this](https://github.com/Jaeyoung-Lim/mavros_controllers), which takes the waypoints and sends attitude commands (I believe), which works with other programmes I am using, with an higher number of waypoints

MihirDharmadhikari, 2022-07-18:
Hi @DavidePatria ,

Sorry for the late reply.
I believe you are using the geometric_controller in the above package and remap the topic published by the planner to the topic [here](https://github.com/Jaeyoung-Lim/mavros_controllers/blob/d3e0fa0e1873cdfdf3233b6777e8f67818137a64/geometric_controller/src/geometric_controller.cpp#L60).
This calculates the control output only when it receives a trajectory message on this topic and only towards the first waypoint. So if you are using this you will have to publish the trajectory accordingly.
Let me know.

Best,
Mihir