ntnu-arl/gbplanner_ros #22: gazebo 9 world process died & librotors_gazebo_multirotor_base_plugin.so: undefined symbol

reem90, 2022-06-24:
I am using Ubuntu 20.04 with ROS Noetic to run this package. I was able to run the ground robot simulation successfully. However, the drone simulation was not. I followed the instructions in the readme file. I installed the models in ./gazebo/models folder.  Here is the output of the terminal. 
```
/gbplanner_node/SensorParams/OS064 0, 0, 0
[INFO] [1656051334.597383, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1656051334.602214, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1656051334.834050629]: Stereo is NOT SUPPORTED
[ INFO] [1656051334.834121696]: OpenGL device: Quadro T2000 with Max-Q Design/PCIe/SSE2
[ INFO] [1656051334.834148694]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1656051335.044938827]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1656051335.046637313]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1656051335.781775331]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1656051335.807447, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1656051368.293094, 875.948000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name rmf_obelix
[ERROR] [1656051368.294511, 875.948000]: Spawn service failed. Exiting.
[ INFO] [1656051368.301889779, 875.948000000]: Physics dynamic reconfigure ready.
[rmf_obelix/spawn_rmf_obelix-12] process has died [pid 515779, exit code 1, cmd /opt/ros/noetic/lib/gazebo_ros/spawn_model -param robot_description -urdf -x 0.0 -y 0.0 -z 0.5 -model rmf_obelix __name:=spawn_rmf_obelix __log:=/home/tii/.ros/log/0dc60e52-f385-11ec-bc0b-586c255f72c4/rmf_obelix-spawn_rmf_obelix-12.log].
log file: /home/tii/.ros/log/0dc60e52-f385-11ec-bc0b-586c255f72c4/rmf_obelix-spawn_rmf_obelix-12*.log
[ INFO] [1656051371.453124837, 875.948000000]: Lidar laser plugin ready, 32 lasers
gzserver: symbol lookup error: /home/tii/gbplanner2_ws/devel/lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev
[gazebo-9] process has died [pid 515762, exit code 127, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/tii/gbplanner2_ws/src/exploration/gbplanner_ros/planner_gazebo_sim/worlds/darpa_subt_final_circuit.world __name:=gazebo __log:=/home/tii/.ros/log/0dc60e52-f385-11ec-bc0b-586c255f72c4/gazebo-9.log].
log file: /home/tii/.ros/log/0dc60e52-f385-11ec-bc0b-586c255f72c4/gazebo-9*.log
```

How can I fix this error 
`gbplanner2_ws/devel/lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev`

and this issue 
```
cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/tii/gbplanner2_ws/src/exploration/gbplanner_ros/planner_gazebo_sim/worlds/darpa_subt_final_circuit.world __name:=gazebo __log:=/home/tii/.ros/log/065c621e-f38b-11ec-ba36-586c255f72c4/gazebo-9.log].
log file: /home/tii/.ros/log/065c621e-f38b-11ec-ba36-586c255f72c4/gazebo-9*.log
```

Kindly assist.

MihirDharmadhikari, 2022-06-27:
Hello @reem90 ,

The second issue that you mention comes from the first issue. Due to the first issue gazebo is unable to launch and that is what the second message says. 
Regarding the first message, can you try setting the `verbose` parameter of the gazebo node to `true` [here](https://github.com/ntnu-arl/gbplanner_ros/blob/95fc8aef0501f2e0f068a8925b65c07ee1953872/gbplanner/launch/rmf/rmf_sim.launch#L34) and send me what it prints out?

Best,
Mihir

mihirk284, 2022-06-27:
This seems to be similar to the issue [here](https://github.com/ethz-asl/rotors_simulator/issues/330).

Can you please share with us the protobuf and libprotobuf versions that you have @reem90 ?

Also try to launch RotorS Simulator after a clean build and let us know if that works.

reem90, 2022-06-27:
> Hello @reem90 ,
>
> The second issue that you mention comes from the first issue. Due to the first issue gazebo is unable to launch and that is what the second message says. Regarding the first [...]

Thank you @MihirDharmadhikari  for your reply. I run `roslaunch gbplanner rmf_sim.launch` with `verbose` is `true`. This is the output without even using the GUI (initilization and start the planner buttons) in the RVIZ 

```
... logging to /home/tii/.ros/log/5a749afe-f5ea-11ec-9e5f-586c255f72c4/roslaunch-reemashour-ThinkPad-P1-Gen-3-46040.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global symbol: pi
when processing file: /home/tii/gbplanner2_ws/src/sim/rotors_simulator/rotors_description/urdf/component_snippets.xacro
included from: /home/tii/gbplanner2_ws/src/sim/rotors_simulator/rotors_description/urdf/rmf_obelix_base.xacro
included from: /home/tii/gbplanner2_ws/src/sim/rotors_simulator/rotors_description/urdf/mav_generic_odometry_sensor.gazebo
included from: /home/tii/gbplanner2_ws/src/sim/rotors_simulator/rotors_description/urdf/rmf_obelix.gazebo
started roslaunch server http://reemashour-ThinkPad-P1-Gen-3:34503/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /gbplanner_node/AdaptiveObbParams/bounding_box_size_max: 35
 * /gbplanner_node/AdaptiveObbParams/local_pointcloud_range: 50.0
 * /gbplanner_node/AdaptiveObbParams/type: kPca
 * /gbplanner_node/BoundedSpaceParams/Global/max_val: [3000.0, 3000.0, ...
 * /gbplanner_node/BoundedSpaceParams/Global/min_val: [-3000.0, -3000.0...
 * /gbplanner_node/BoundedSpaceParams/Global/type: kCuboid
 * /gbplanner_node/BoundedSpaceParams/Local/max_extension: [20.0, 20.0, 20.0]
 * /gbplanner_node/BoundedSpaceParams/Local/max_val: [15.0, 15.0, 3.0]
 * /gbplanner_node/BoundedSpaceParams/Local/min_extension: [-20.0, -20.0, -2...
 * /gbplanner_node/BoundedSpaceParams/Local/min_val: [-15.0, -15.0, -3.0]
 * /gbplanner_node/BoundedSpaceParams/Local/type: kCuboid
 * /gbplanner_node/BoundedSpaceParams/LocalAdaptiveExp/max_val: [10.0, 10.0, 0.75]
 * /gbplanner_node/BoundedSpaceParams/LocalAdaptiveExp/min_val: [-10.0, -10.0, -0...
 * /gbplanner_node/BoundedSpaceParams/LocalAdaptiveExp/type: kCuboid
 * /gbplanner_node/BoundedSpaceParams/LocalSearch/max_val: [50.0, 50.0, 1.0]
 * /gbplanner_node/BoundedSpaceParams/LocalSearch/min_val: [-50.0, -50.0, -1.0]
 * /gbplanner_node/BoundedSpaceParams/LocalSearch/type: kCuboid
 * /gbplanner_node/NoGainZones/g1/max_val: [2.0, 10.0, 10.0]
 * /gbplanner_node/NoGainZones/g1/min_val: [-50.0, -10.0, -1...
 * /gbplanner_node/NoGainZones/g1/type: kCuboid
 * /gbplanner_node/PlanningParams/augment_free_voxels_time: 3
 * /gbplanner_node/PlanningParams/auto_homing_enable: True
 * /gbplanner_node/PlanningParams/cluster_vertices_for_gain: True
 * /gbplanner_node/PlanningParams/clustering_radius: 0.5
 * /gbplanner_node/PlanningParams/edge_length_max: 2.0
 * /gbplanner_node/PlanningParams/edge_length_min: 0.2
 * /gbplanner_node/PlanningParams/edge_overshoot: 0.0
 * /gbplanner_node/PlanningParams/exp_gain_voxel_size: 0.8
 * /gbplanner_node/PlanningParams/exp_sensor_list: ['OS064']
 * /gbplanner_node/PlanningParams/free_frustum_before_planning: False
 * /gbplanner_node/PlanningParams/free_voxel_gain: 0.0
 * /gbplanner_node/PlanningParams/freespace_cloud_enable: False
 * /gbplanner_node/PlanningParams/geofence_checking_enable: True
 * /gbplanner_node/PlanningParams/go_home_if_fully_explored: True
 * /gbplanner_node/PlanningParams/leafs_only_for_volumetric_gain: True
 * /gbplanner_node/PlanningParams/nearest_range: 2.0
 * /gbplanner_node/PlanningParams/nearest_range_max: 2.0
 * /gbplanner_node/PlanningParams/nearest_range_min: 0.5
 * /gbplanner_node/PlanningParams/nonuniform_ray_cast: True
 * /gbplanner_node/PlanningParams/num_edges_max: 9000
 * /gbplanner_node/PlanningParams/num_loops_cutoff: 2000
 * /gbplanner_node/PlanningParams/num_loops_max: 4000
 * /gbplanner_node/PlanningParams/num_vertices_max: 700
 * /gbplanner_node/PlanningParams/occupied_voxel_gain: 0.0
 * /gbplanner_node/PlanningParams/path_direction_penalty: 0.1
 * /gbplanner_node/PlanningParams/path_interpolation_distance: 0.5
[... 226 lines cut ...]
[ INFO] [1656329035.312244683]: Physics dynamic reconfigure ready.
[INFO] [1656329035.434293, 0.000000]: wait_for_service(/gazebo/spawn_urdf_model): finally were able to contact [rosrpc://reemashour-ThinkPad-P1-Gen-3:50139]
[INFO] [1656329035.435762, 0.000000]: Calling service /gazebo/spawn_urdf_model
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[Wrn] [ColladaLoader.cc:1441] emission texture not supported
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
...
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported
[Wrn] [ColladaLoader.cc:1928] Triangle input semantic: 'COLOR' is currently not supported

[Err] [Plugin.hh:212] Failed to load plugin librotors_gazebo_ros_interface_plugin.so: /home/tii/gbplanner2_ws/devel/lib/librotors_gazebo_ros_interface_plugin.so: undefined symbol: _ZTIN11gz_std_msgs7Float32E
[... cut ...]
[gazebo-8] process has died [pid 46123, exit code 127, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver --verbose -e ode /home/tii/gbplanner2_ws/src/exploration/gbplanner_ros/planner_gazebo_sim/worlds/darpa_subt_final_circuit.world __name:=gazebo __log:=/home/tii/.ros/log/5a749afe-f5ea-11ec-9e5f-586c255f72c4/gazebo-8.log].
log file: /home/tii/.ros/log/5a749afe-f5ea-11ec-9e5f-586c255f72c4/gazebo-8*.log
[gbplanner_ui-16] process has finished cleanly
log file: /home/tii/.ros/log/5a749afe-f5ea-11ec-9e5f-586c255f72c4/gbplanner_ui-16*.log

```

reem90, 2022-07-05:
@MihirDharmadhikari @mihirk284 Any update?

reem90, 2022-08-30:
@MihirDharmadhikari There was a conflict with another plugin from a different package. I kept only this package in my workspace, and I was able to run the flying robot simulation.

closed 2022-08-30 by reem90