ntnu-arl/gbplanner_ros #24: Exploration Demo Not Working

SwiftGust, 2022-06-29:
Hi, 
I have encountered issue when running demo simulations on ROS Melodic (Ubuntu 18.04) following instructions
After clicking "Initialization" and "Start Planner" in order,
the exploration planner never leads vehicle to out of starting location.
It just goes back and forth in starting area
This behavior is similar in case of  'smb_sim' too.

How can I make exploration running? 

Only console output to note for both cases
`[GBPLANNER] Very low local gain. Triggering global planner`

I'm using latest gbplanner2 branch of this repository, 
resolved few compile issue other than that it is default as is.

- Case - `$ roslaunch gbplanner rmf_sim.launch` after

[image]

- Case - `$ roslaunch gbplanner smb_sim.launch` after 400 secs of run

[image]

MihirDharmadhikari, 2022-06-30:
Seems like the planner is not able to build the graph. Are you using the default parameters? Also, had you built the code with `-DCMAKE_BUILD_TYPE=Release`?

SwiftGust, 2022-07-04:
Hi, @MihirDharmadhikari 
sorry for late response 
I have built code with ` -DCMAKE_BUILD_TYPE=RelWithDebInfo` 
Is this could cause a trouble? then which mode should I have to build code?

MihirDharmadhikari, 2022-07-23:
Hello,

I tried to reproduce this error from my side on a clean installation but was unable to reproduce this. 
You can try one thing, which is setting this to `Verbosity::DEBUG` and see if there are any warnings or errors printed by the planner.

Best,
Mihir