ntnu-arl/gbplanner_ros #26: Change all submodules git repos url due to github's new requirments on security

nvandamme, 2022-08-01:
https://github.com/ntnu-arl/gbplanner_ros/blob/ca1279152085e945724dc730f9d77a78e314a365/packages_ssh.rosinstall#L4

All git@github.com ssh URLs should be changed with their https equivalents due to new security policies

nkhedekar, 2022-08-03:
Can you please post a link to the new security policy that you are referring to?

nvandamme, 2022-08-05:
Hi,

Sorry for the delay and the short message ealier.

Because of this new policy : https://github.blog/2020-12-15-token-authentication-requirements-for-git-operations/
And, with SSH based git operation, the requirement of using an SSH key : for automated operations, CI/DI, using SSH based urls require human operations as they cannot be executed anonymously (via HTTPS).

Thus, it would be really nice if ssh based urls could be remapped with https based urls for git repos.

Anyway thanks for your excellent work !

nkhedekar, 2022-08-05:
Thanks for the link and explanation! I would propose that we don't change the ssh urls since this is the primary method that everyone has been using so far (and can continue using if they would like) but instead create a new `packages_https.rosinstall` that is maintained in parallel but with links that use HTTPS. The new file can then be used for CI/CD by setting the appropriate tokens by the user.

nkhedekar, 2022-08-05:
A PR on this would be welcome :smile:

MihirDharmadhikari, 2022-08-12:
Hello @nvandamme ,
I have created a packages_https.rosinstall for https based URLs of packages in [this](https://github.com/ntnu-arl/gbplanner_ros/tree/feature/https_rosinstall) branch. Let me know if this is what you mentioned.

Best,
Mihir

MihirDharmadhikari, 2022-08-27:
Resolved with PR #27

closed 2022-08-27 by MihirDharmadhikari