ntnu-arl/gbplanner_ros #30: Plan to Waypoint Function cannot work

Rye450, 2022-10-03:
Hello,I am trying the Plan to Waypoint function in the UI which you mentioned in here[https://github.com/ntnu-arl/gbplanner_ros/wiki/Demo#5-plan-to-waypoint-using-the-global-graph].
I followed your instructions:First,I set the goal point using the 2D nav goal button in rviz interface.And then,I clicked the Plan to Waypoint button in the UI.
From the terminal output,I can see that the goal point has been set.However,the planning path doesn't go to goal point.
[image]
[image]
Could you tell why?

MihirDharmadhikari, 2022-10-10:
Hi @Rye450 ,

To send the robot to a specific point, the robot must have first mapped some area around it. Then you can give the waypoint using the following steps:
- Press Stop Planner
- Give the waypoint with the 2D nav goal
- Press Plan to Waypoint.
If Stop Planner is not pressed, then the planner will not go into the waypoint following mode.
If this does not work, can you compile the code with [this](URL) set to `DEBUG` and then rerun it? This should print out if there is any error.
Let me know if this works.

Best regards,
Mihir

Rye450, 2022-10-12:
Hi @MihirDharmadhikari 
It works when I followed your instructions as above.I guess I missed the Press Stop Planner step before.
Thanks a lot for sharing this awesome work!

MihirDharmadhikari, 2022-10-22:
Glad to hear that it worked!

closed 2022-10-22 by MihirDharmadhikari