ntnu-arl/gbplanner_ros #31: gbplanner_node crash, planner service failed.

antoan-flox, 2022-12-02:
Hello, 

I built the package on ROS Noetic and experienced a crash, here is the output:

`roslaunch gbplanner smb_sim.launch
... logging to /home/antoan/.ros/log/c189b720-726c-11ed-9f3d-035f770cbc42/roslaunch-Ronin-736510.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Ronin:38123/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /gbplanner_node/AdaptiveObbParams/bounding_box_size_max: 35
 * /gbplanner_node/AdaptiveObbParams/local_pointcloud_range: 50.0
 * /gbplanner_node/AdaptiveObbParams/type: kPca
 * /gbplanner_node/BoundedSpaceParams/Global/max_val: [1000.0, 1000.0, ...
 * /gbplanner_node/BoundedSpaceParams/Global/min_val: [-1000.0, -1000.0...
 * /gbplanner_node/BoundedSpaceParams/Global/type: kCuboid
 * /gbplanner_node/BoundedSpaceParams/Local/max_extension: [20.0, 20.0, 20.0]
 * /gbplanner_node/BoundedSpaceParams/Local/max_val: [15.0, 15.0, 3.0]
 * /gbplanner_node/BoundedSpaceParams/Local/min_extension: [-20.0, -20.0, -2...
 * /gbplanner_node/BoundedSpaceParams/Local/min_val: [-15.0, -15.0, -3.0]
 * /gbplanner_node/BoundedSpaceParams/Local/type: kCuboid
 * /gbplanner_node/BoundedSpaceParams/LocalAdaptiveExp/max_val: [10.0, 10.0, 0.75]
 * /gbplanner_node/BoundedSpaceParams/LocalAdaptiveExp/min_val: [-10.0, -10.0, -0...
 * /gbplanner_node/BoundedSpaceParams/LocalAdaptiveExp/type: kCuboid
 * /gbplanner_node/BoundedSpaceParams/LocalSearch/max_val: [50.0, 50.0, 10.0]
 * /gbplanner_node/BoundedSpaceParams/LocalSearch/min_val: [-50.0, -50.0, -1...
 * /gbplanner_node/BoundedSpaceParams/LocalSearch/type: kCuboid
 * /gbplanner_node/GeofenceParams/AreaList: ['GLeft', 'GBack'...
 * /gbplanner_node/GeofenceParams/GBack/center: [-3.0, 0.0, 0.0]
 * /gbplanner_node/GeofenceParams/GBack/size: [2.0, 6.0, 0.0]
 * /gbplanner_node/GeofenceParams/GLeft/center: [0.0, 4.0, 0.0]
 * /gbplanner_node/GeofenceParams/GLeft/size: [6.0, 2.0, 0.0]
 * /gbplanner_node/GeofenceParams/GRight/center: [0.0, -4.0, 0.0]
 * /gbplanner_node/GeofenceParams/GRight/size: [6.0, 2.0, 0.0]
 * /gbplanner_node/NoGainZones/g1/max_val: [2.0, 50.0, 20.0]
 * /gbplanner_node/NoGainZones/g1/min_val: [-50.0, -50.0, -2...
 * /gbplanner_node/NoGainZones/g1/type: kCuboid
 * /gbplanner_node/PlanningParams/augment_free_voxels_time: 1
 * /gbplanner_node/PlanningParams/auto_global_planner_enable: True
 * /gbplanner_node/PlanningParams/auto_homing_enable: True
 * /gbplanner_node/PlanningParams/bound_mode: kExtremeBound
 * /gbplanner_node/PlanningParams/cluster_vertices_for_gain: True
 * /gbplanner_node/PlanningParams/clustering_radius: 0.5
 * /gbplanner_node/PlanningParams/edge_length_max: 3.0
 * /gbplanner_node/PlanningParams/edge_length_min: 0.1
 * /gbplanner_node/PlanningParams/edge_overshoot: 0.0
 * /gbplanner_node/PlanningParams/exp_gain_voxel_size: 0.8
 * /gbplanner_node/PlanningParams/exp_sensor_list: ['VLP16']
 * /gbplanner_node/PlanningParams/free_frustum_before_planning: False
 * /gbplanner_node/PlanningParams/free_voxel_gain: 0.0
 * /gbplanner_node/PlanningParams/freespace_cloud_enable: False
 * /gbplanner_node/PlanningParams/geofence_checking_enable: True
 * /gbplanner_node/PlanningParams/global_frame_id: world
 * /gbplanner_node/PlanningParams/global_path_inclination_check: True
 * /gbplanner_node/PlanningParams/go_home_if_fully_explored: True
 * /gbplanner_node/PlanningParams/hanging_vertex_penalty: 5.0
 * /gbplanner_node/PlanningParams/homing_backward: False
 * /gbplanner_node/PlanningParams/interpolate_projection_distance: False
 * /gbplanner_node/PlanningParams/leafs_only_for_volumetric_gain: False
 * /gbplanner_node/PlanningParams/max_ground_height: 0.7
 * /gbplanner_node/PlanningParams/max_inclination: 0.4363323129985824
 * /gbplanner_node/PlanningParams/nearest_range: 2.5
 * /gbplanner_node/PlanningParams/nearest_range_max: 3.0
[... 367 lines cut ...]
    @     0x7f0eae22d859 abort
    @     0x7f0eae22d729 (unknown)
    @     0x7f0eae23efd6 __assert_fail
    @     0x7f0eaf47bc11 Eigen::DenseCoeffsBase<>::operator()()
    @     0x7f0eaf47a234 _ZN17MapManagerVoxbloxIN7voxblox10TsdfServerENS0_9TsdfVoxelEE13getScanStatusERN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEERSt6vectorIS6_SaIS6_EERSt5tupleIJiiiEERS8_ISt4pairIS6_N10MapManager11VoxelStatusEESaISI_EER16SensorParamsBase
    @     0x7f0eafbc3017 explorer::Rrg::computeVolumetricGainRayModel()
    @     0x7f0eafbba192 explorer::Rrg::expandGlobalGraphTimerCallback()
    @     0x7f0eafc310fb boost::_mfi::mf1<>::operator()()
    @     0x7f0eafc29371 boost::_bi::list2<>::operator()<>()
    @     0x7f0eafc2229d boost::_bi::bind_t<>::operator()<>()
    @     0x7f0eafc1ab2e boost::detail::function::void_function_obj_invoker1<>::invoke()
    @     0x7f0eaff07608 ros::TimerManager<>::TimerQueueCallback::call()
    @     0x7f0eaff29172 ros::CallbackQueue::callOneCB()
    @     0x7f0eaff2a883 ros::CallbackQueue::callAvailable()
    @     0x7f0eaff7dfcf ros::SingleThreadedSpinner::spin()
    @     0x7f0eaff6621f ros::spin()
    @     0x557a276b8a3a main
    @     0x7f0eae22f083 __libc_start_main
    @     0x557a276b871e _start
[ERROR] [1670004946.273309155, 477.380000000]: Planner service failed
[gbplanner_node-17] process has died [pid 736674, exit code -6, cmd /home/antoan/dev/gbplanner2_ws/devel/lib/gbplanner/gbplanner_node odometry:=/ground_truth/state /pointcloud:=/input_pointcloud __name:=gbplanner_node __log:=/home/antoan/.ros/log/c189b720-726c-11ed-9f3d-035f770cbc42/gbplanner_node-17.log].
log file: /home/antoan/.ros/log/c189b720-726c-11ed-9f3d-035f770cbc42/gbplanner_node-17*.log
[ INFO] [1670004946.806100546, 477.890000000]: Planning iteration 9
[ERROR] [1670004946.806156020, 477.890000000]: Planner service failed
`

MihirDharmadhikari, 2022-12-09:
Hi @antoan-flox ,

Did you experience this issue persistently or was this happening only occasionally?

Best,
Mihir

MihirDharmadhikari, 2023-05-25:
Hi @yhy130531,

Thanks for letting us know. I am investigating this issue and will get back to you on this.

yhy130531, 2023-05-25:
> Hi @yhy130531,
> 
> Thanks for letting us know. I am investigating this issue and will get back to you on this.

Thank you very much. One thing that needs to be explained is that our emulator does not use Gazebo, but only uses Rviz display. The drone control, odometer, and point cloud generation are all ours. I only tested the exploration algorithm. When I start the algorithm , the above error occurred.

yhy130531, 2023-05-25:
> 
Thank you very  much @MihirDharmadhikari. One thing that needs to be explained is that our emulator does not use Gazebo, but only uses Rviz display. The drone control, odometer, and point cloud generation are all ours. I only tested the exploration algorithm. When I start the algorithm , the above error occurred.

> Hi @yhy130531,
> 
> Thanks for letting us know. I am investigating this issue and will get back to you on this.

MihirDharmadhikari, 2023-05-25:
The planner can work with a custom setup as long as the appropriate topics for the point cloud and odometry are set and a position controller for tracking the output path is available. So I think your setup should be fine. 
Just out of curiosity, did you encounter the same error with our simulation setup as well? This might help narrow down the issue.

yhy130531, 2023-05-25:
> The planner can work with a custom setup as long as the appropriate topics for the point cloud and odometry are set and a position controller for tracking the output path is available. So I think [...]

  Hello, @MihirDharmadhikari ,I still have a question, whether the lidar must be 64 lines in the simulation algorithm or the real experiment. Currently, the lidar used in our simulation is only OS128, VLP32, Mid360, Avia, etc.

MihirDharmadhikari, 2023-05-25:
The LiDAR need not have 64 beams. The planner can take any point cloud as input as long as it is in the sensor frame so any other LiDAR simulator should be fine.

yhy130531, 2023-05-25:
> 

Copy that, thanks!

SFakoorian0111, 2023-11-28:
@MihirDharmadhikari I see the same issue when I run your sim example ` roslaunch gbplanner smb_sim.launch`
I attached a screen shot but this issue happens when I click on "Start Planner". When I click the "Initialization" the robot moves for a couple of steps and then it stops. 
[image]

MihirDharmadhikari, 2023-11-29:
Hello,

This might be related to issue #20. We are investigating the root cause of it. Meanwhile, you can set [this](https://github.com/ntnu-arl/gbplanner_ros/blob/bc968146d0fa0a1dbc0f2e7c4667a54d2226ff5e/planner_common/CMakeLists.txt#L20) parameter in the 'planner_common' package to `0`, clean the package, and rebuild. However, note that this will use a more conservative volumetric gain computation.

Best,
Mihir

Ray005, 2024-12-02:
I encountered the same issue during simulation. Ultimately, I found that using ```catkin build``` instead of ```catkin_make``` to compile the project resolved the problem. Additionally, the code runs faster when compiled with ```catkin build```. 

Perhaps others facing this issue can consider checking the building command.

MihirDharmadhikari, 2024-12-02:
Hi @Ray005,

We highly recommend using `catkin build` and not `catkin_make` for building the project. Additionally, the project should be built in `release` mode. The exact commands to built the project can be found in the readme of this repository or in the [wiki](https://github.com/ntnu-arl/gbplanner_ros/wiki/Installation#build).

Best,
Mihir