ntnu-arl/gbplanner_ros #34: no legged robot

Shai2022, 2023-05-13:
Hello,
Does this repository also support ground robots without legs?
Thanks

MihirDharmadhikari, 2023-05-14:
Hello @Shai2022,

Yes, the planner can be used for other ground robots as well. However, you will need appropriate path follower for your robot.

Best,
Mihir

Shai2022, 2023-05-27:
I apologize for the misunderstanding, but could you please provide more details about what i need to implement ?

MihirDharmadhikari, 2023-05-28:
Hello,

The output of the planner is a trajectory of the ROS message type MultiDOFJointTrajectory which is published on the topic `/<robot_name>/command/trajectory`. You will need to implement a controller for your robot that can subscribe to this trajectory and follow it.
Let me know if this helps.

Best,
Mihir

DhruvSinghiitmandi, 2024-01-09:
Hi @MihirDharmadhikari 
Could  you please tell more on writing the controller that you speak about here?
"Hello,

The output of the planner is a trajectory of the ROS message type MultiDOFJointTrajectory which is published on the topic /<robot_name>/command/trajectory. You will need to implement a controller for your robot that can subscribe to this trajectory and follow it."

MihirDharmadhikari, 2024-01-10:
Hello @DhruvSinghiitmandi ,

The controller depends on your robot. For the differential drive robot that we provide in the demo, we use a simple pure pursuit path tracker. You can find the implementation here: https://github.com/ntnu-arl/smb_simulator/tree/main/smb_path_tracker.
Note that this tracker uses a ROS Action interface, but the underlying controller remains the same even if you use `kTopic` or `kAction` in the planner, only the interface changes.

I hope this helps.

Best,
Mihir