ntnu-arl/gbplanner_ros #35: When homing, the robot doesn't return exactly to the starting position.

fantasycenturay, 2023-06-02:
I assume the position after initialization is the starting position for the planner. The robot will only return to a position approximately half to one meter down the path from the starting position. I guess that's where the first node of the planning graph is and that the starting position is not part of the planning graph, thus not on the homing path?
I will look into the code. In the meantime, if someone has any idea of the cause, please let me know.

fantasycenturay, 2023-06-02:
Apparently, the homing position is set using `Rrg::setHomingPos() `, which is called back by "gbplanner/set_homing_pos", which eventually leads to pci services "pci_set_homing_pos" and "planner_control_interface/std_srvs/set_homing_position_here". However, I can't find the clients for the last two services.
Any idea of that? Or how exactly is the setHomingPos() function used?

MihirDharmadhikari, 2023-06-21:
Hi,

Sorry for the late reply. The homing position is the position from where the first planning iteration was triggered. This is where the root vertex of the global graph is and it is used as the default home location. 
The `planner_control_interface/std_srvs/set_homing_position_here` service is used to change the home position to the position where the robot is.
I hope this helps.

Best,
Mihir

fantasycenturay, 2023-06-26:
Thanks Mihir for clarifying this.
I tried the set_homing_position_here service the other day and it did seem to have changed the home position. I also changed the goal_reaching_threshold value, but need to fix my tf_tree before testing how accurate it is.

Btw, is the initialization procedure necessary for the gbplanner, or is it just there to prepare the robot?

MihirDharmadhikari, 2023-06-26:
Hi @fantasycenturay,

Glad to hear that it worked. 

The initialization procedure is necessary to clear some space around the robot in the volumetric map so that the robot is in free space (if initialization motion is not done it will be partly in the unknown space) when the planner is triggered and the planner can build the graph. This need not be exactly as done in the demo. For aerial robots, we have found that takeoff to a certain height (depending on your application and robot), moving down a little (about the distance equal to the robot height), and then moving forward (or in any direction in x-y plane for that matter) a little work the best. However, just taking off and moving forward works as well.

Best,
Mihir

fantasycenturay, 2023-06-27:
@MihirDharmadhikari 
That explains it. Btw we are working on a ground robot to do basically the same thing you guys achieved years ago. Thumbs up to all of you for sharing this awesome work and help the rest of us get started.
Best,
Qiang

closed 2023-06-27 by fantasycenturay

MihirDharmadhikari, 2023-06-27:
Thank you very much for your encouraging words! All the best for your project!