ntnu-arl/gbplanner_ros #37: Build bug

lyf23333, 2023-07-14:
Hi, thank you for sharing this great work! I would like to try on my machine, but I keep getting error when I run catkin build. After I searched, this should be a error due to C++, so I ran catkin build -DCMAKE_CXX_STANDARD=17. Still showing the same error. DO you have any idea why is that? Appreciate your help! 

[image]

closed 2023-07-16 by lyf23333

A-Nobel, 2024-11-12:
Hello! Would you like to share the solution? Thanks a lot!

A-Nobel, 2024-11-12:
add_definitions(-std=c++14) -> add_definitions(-std=c++17)
[image]