ntnu-arl/gbplanner_ros #40: Is it necessary to have the "max_ground_height"

fantasycenturay, 2023-08-01:
As explained in the following code comment, it seems that the "max_ground_height" param is there to project "ground_height", "groud_dist" and "robot_height" to the same reference level surface. Why is this necessary?
`      // Get the point on the ground at a heigh of
      // planning_params_.max_ground_height
      projected_edge_pt(2) -=
          (ground_height - planning_params_.max_ground_height);`

MihirDharmadhikari, 2023-08-09:
Sorry for the late reply.
The `max_ground_height` is the height above the ground at which a vertex in the graph is projected to get before doing the collision checking.

fantasycenturay, 2023-09-01:
Thanks for the explanation. I tested it on flat surface, it worked as explained and expected. I will do more tests on uneven surfaces today.

closed 2023-09-01 by fantasycenturay