ntnu-arl/gbplanner_ros #41: Using GBP in our own uav simulator

chrisuzuki62, 2023-08-09:
Hi I am currently trying to get the GBP Planner to work in our own UAV simulator and I am running into a couple of issues.
We have our own controller so I am converting the trajectory message for our controller.

1. Does it matter if I use smb or rmf configs for gbp? (I am using and editing the original gbp launch file but removed the gazebo portion).
2. I am using our own UAV simulator and pass in the odometry topic and point cloud topic into the planner (using a realsense camera for pc). Am I missing any other parameters I should changing?

3. The trajectory generated is only where the drone is spawned and never allows the drone to progress forward.
Why is this happening?

Any help would be greatly appreciated. 
Attached below are my two launch files:
[sim1.launch.pdf](https://github.com/ntnu-arl/gbplanner_ros/files/12297943/sim1.launch.pdf)
[sim2.launch.pdf](https://github.com/ntnu-arl/gbplanner_ros/files/12297944/sim2.launch.pdf)

MihirDharmadhikari, 2023-08-09:
Hello @chrisuzuki62 ,

1. I would recommend using the rmf config and launch files as there are some parameters that need to be different for a ground and an aerial robot. In principle the planner works the same but it will be easier to get the planner working with the config files for the same type of robot.
2. Only those two topics are necessary for the planner. However, you need to ensure that there is a tf between the sensor frame in which the point cloud is and the fixed frame (usually `world`).
3. It is possible that the planner is unable to calculate the path hence the only waypoint in the path is the current state. Can you send me a video of the behavior along with the terminal log? Try to visualize the voxblox map (topic is `gbplanner_node/occupied_nodes`)

The launch files look fine to me.
Let me know if this helps.

Best,
Mihir

chrisuzuki62, 2023-08-10:
Hello @MihirDharmadhikari,

Thank you for the quick reply.
I attached the video of the current results I am seeing running GBP.
I am launching the planner using sim1.launch from the previous attachments.
Is it because the uav cannot see any obstacles in front of it?

Thanks,
Chris

https://github.com/ntnu-arl/gbplanner_ros/assets/97257100/6f34a6ed-29c4-4aef-a4d8-56808a77feab

chrisuzuki62, 2023-08-16:
Hi @MihirDharmadhikari,
I wanted to follow up that unless I manually move the drone in my simulation so more of the walls of the map are included within the Voxel Map the planner will not generate a trajectory.

chrisuzuki62, 2023-08-16:
Hi @MihirDharmadhikari,

Does the Initialization use the same trajectory command?
Also I noticed the exploration tends to come to explore a new region and comeback to the start and go out and explore a new region.

MihirDharmadhikari, 2023-08-21:
Yes, it uses the same trajectory command.
Regarding your comment about the exploration behavior, can you send me an image of the environment to understand how it looks?

chrisuzuki62, 2023-09-06:
Hi @MihirDharmadhikari 
Here is the image you asked for
[image]

chrisuzuki62, 2023-09-06:
Hi @MihirDharmadhikari 
Another question I had was how should I change the sensor params for a realsense camera. 
I am a bit confused with the setting since they seem to be made for a lidar.
SensorParams:
  sensor_list:      ["OS064"]
  OS064:
    type:           kLidar
    max_range:      20.0
    center_offset:  [0.0, 0.0, 0.0]
    rotations:      [0.0, 0.0, 0.0]
    fov:            [rad(0.785), rad(-0.785)]
    resolution:     [rad(7.0*pi/180), rad(7.0*pi/180)]
    frontier_percentage_threshold: 0.04

MihirDharmadhikari, 2023-09-06:
Hi @chrisuzuki62 ,

Regarding the sensor params, even if you keep the sensor/type as `kLidar` it should be fine. It uses a cylindrical sensor frustum model which is slightly different than what a depth camera has but for the purposes of volumetric gain calculation, it is not a problem. 
You mainly need to change the following parameters:
- `rotations`: This is the orientation of the sensor w.r.t. the frame of the body frame of the robot that the odometry solution is providing (e.g. in the demos it is the `base_link`).
- `fov`: Set this to that of the sensor

Regarding your question about the planner failing, did you try moving the robot to the mapped area before triggering the planner? As I mentioned last time, it seems that when you trigger the planner, the robot is in the unmapped area hence the planner is not able to plan a path. If your sensor is in the front of the robot, then simply doing the initialization motion (by clicking on the `Initialization` button in the UI) will work.

Let me know if this works.

Best,
Mihir

chrisuzuki62, 2023-09-06:
Hi @MihirDharmadhikari ,

I fixed the issue of the initialization not working and drone now does the initialization motion, however afterwards when I hit start planning the planner will fail if I change the fov.
Below is the screenshot of the error with the fov that I used. I also tried fov I mentioned in my earlier post and it failed as well.

[image]

MihirDharmadhikari, 2023-09-06:
Can you send a few lines that were printed in the terminal above the `ros::ServiceCallbackHelper<>T::call()`? I want to check which function it failed in.

MihirDharmadhikari, 2023-09-07:
Hi @chrisuzuki62 ,

I have seen this issue occurring before. One solution, for now, will be to set [this](https://github.com/ntnu-arl/gbplanner_ros/blob/5f530328e5c84334ff52326d5c57b8f53db571d0/planner_common/CMakeLists.txt#L20) parameter to `0` and clean and rebuild the `planner_common` and `gbplanner` packages. This changes the ray casting method to a basic one which won't have this problem. 
Let me know if this works.

Best,
Mihir

chrisuzuki62, 2023-09-07:
Hi @MihirDharmadhikari,

Thank you I got the planner to work with different fovs now.
I notice however the planner still does not rotate the drone around as much when exploring to face camera to can unknown area.
Also what units are the boundingspace params in. I would like to edit them so the I can test the planner on different sized maps.
And how should I edit these params to best test your planner with different maps and within range of our drones sims camera.