ntnu-arl/gbplanner_ros #42: How to save the map.into file?

Zheng-CW, 2023-12-06:
Hello, sincerely thank you for your work, very useful to me! I use the robot to explore and create a map, how do I save this map data into the file? Does your algorithm provide a function to save the map data? Looking forward to your suggestions! Thanks again for your contributions!

MihirDharmadhikari, 2023-12-06:
Hello @Zheng-CW,

If you want to save the voxblox map there is a service `/gbplanner_node/save_map` that you can call. It is of type `voxblox_msgs/FilePath`. You only need to provide the path to the map. Similarly, there is a service to load the saved voxblox map as well `/gbplanner_node/load_map`.

Best,
Mihir

Zheng-CW, 2023-12-07:
Thank you very much for your suggestions! When I save the map data through the rosservice call /gbplanner_node/save_map /home/zcw/local_map /map1.bt command, no matter I save the map several times during the robot exploration process or execute the command only once after the exploration is over, It ends up being the same amount of data! Is the effect of both methods the same? Can it be used to save maps?  Glad and thank you for answering my questions!