ntnu-arl/gbplanner_ros #52: Setting a goal for planner

celikemir, 2024-06-01:
Hi @mihirk284 thank you for great work.

I wonder that is it possible to set a navigation point for planner like geometric location ( lat, lon ) and stoping the planner when arrive ?

MihirDharmadhikari, 2024-06-04:
Hello @celikemir,

As gbplanner is an exploration planner it does not accept a goal to navigate to in unmapped space. However, if you have the robot explore and then want to plan to a point in the mapped space, you can use the functionality explained [here](https://github.com/ntnu-arl/gbplanner_ros/wiki/Demo#5-plan-to-waypoint-using-the-global-graph).

Best,
Mihir

MihirDharmadhikari, 2024-06-10:
Hello,

There is a function in the code [here](https://github.com/ntnu-arl/gbplanner_ros/blob/77afce521a3aa7a4b94bf6a43a3c88ec08b6d544/gbplanner/src/rrg.cpp#L2189) that accepts a goal point and can plan a path to it in the known map. You can use this function for your task.

Best,
Mihir