ntnu-arl/gbplanner_ros #53: How to prevent collision

celikemir, 2024-06-09:
Hi,

I have a somewhat challenging map where the elevation changes. In some areas, the elevation from the ground is really small, like 5 cm or below. How can I prevent the planner from drawing paths in these kinds of places? There are pavements, and the planner draws above the pavements, which it should not do. How can I prevent this behavior? I modified col_check = 2 in the CMakeLists file, which resulted in some improvement but didn't fully solve the problem.
[image]

MihirDharmadhikari, 2024-06-10:
Hello,

The pavements are tricky to tackle. Due to the size of the voxels in the voxblox map, the planner won't be able to differentiate between a slope and a pavement that is one voxel size tall. One workaround to this is to reduce the voxel size (currently, it is 0.2 you can make it 0.1m). However, keep in mind that this will increase the memory significantly as well as increase the planning time.

Best,
Mihir

celikemir, 2024-06-10:
thank you Mihir. The global area, the red sqaure, is not cover all my map so I have modify the gbplanner.yaml [there](https://github.com/ntnu-arl/gbplanner_ros/blob/77afce521a3aa7a4b94bf6a43a3c88ec08b6d544/gbplanner/config/smb/gbplanner_config.yaml#L34) is a defined number even though I changed so small or large number cant see difference dimension of the red box.

celikemir, 2024-06-10:
yeah it brings realy huge memory consumption. maybe you can adress me reduce voxblock count I tried to reduce with some yaml[https://github.com/ntnu-arl/gbplanner_ros/blob/77afce521a3aa7a4b94bf6a43a3c88ec08b6d544/gbplanner/config/smb/voxblox_sim_config.yaml#L4](url)

Can you tell me some of the parameters meaning ? thank you

celikemir, 2024-06-12:
@MihirDharmadhikari How to increase per voxblock size to prevent collusion there is huge difference between planned path hegiht and obstacle however robot suppose to go over from obstackle but it overturn