ntnu-arl/gbplanner_ros #56: Paths generated are way too close to edges

Josgonmar, 2024-07-16:
Hi, thanks a lot for your awesome work!

I've been using your package to explore some sort of warehouse with corridors and stands, and I noticed how in some cases the paths generated are too close to edges when changing from one corridor to another. It seems that the path improvement fails to provide a further path even though there is enough free space to do so:

[image]

Is there a work around to avoid these type of things? I was thinking about some sort of inflation for the map, but the only thing I managed to do for better results was just to increase the size of the voxels, but it makes some corridors too narrow to be explored.

Thanks in advance!

tiralonghipol, 2024-09-21:
hei @Josgonmar ! you can inflate the robot size and the safe collision extension in the config file [here](https://github.com/ntnu-arl/gbplanner_ros/blob/7301b5350915cf3a4055072ad559842db37b4ca9/gbplanner/config/rmf_obelix/gbplanner_config.yaml#L9)

Josgonmar, 2024-09-23:
Hey @tiralonghipol, thanks for the reply!

I did play a little bit with the size extension and similar params, but i was still getting the same result, usually when the path had to turn around an edge.
What i did in the end in order to prevent this from happening so often was to divide the path in 2 segments where the turn was made, and improving the first one. Then, after adding the remaining segment to the improved one, i call the same function again with the whole path. This way i usually get better results, as the increased computation is not a problem for me :smile:

I'm closing this issue now!

closed 2024-09-23 by Josgonmar