ntnu-arl/gbplanner_ros #59: Integration with CMU development Environment

ShayonMitra, 2025-02-05:
Integration with cmu-exploration: Guidance Needed

I'm attempting to integrate this planner with the cmu-exploration development environment. I'm encountering some challenges and would appreciate guidance on the process.  Could you please point me towards any relevant documentation or examples?  Information about appropriate launch files to use as a starting point would be particularly helpful.

ShayonMitra, 2025-02-06:
Hi @MihirDharmadhikari 
Thanks for the reply. I have written this launch file. I am launching this alongside system_indoor.launch(the simulator's file). 
`<launch>

 <arg name="robot_name" default="smb"/>
  <arg name="gazebo_gui_en" default="true"/>
  <arg name="use_sim_time" default="true"/>
  <arg name="rviz_en" default="true" />
  <arg name="launch_prefix" default=""/> 
  <param name="use_sim_time" value="$(arg use_sim_time)"/>

  
  <arg name="gbplanner_config_file" default="$(find gbplanner)/config/$(arg robot_name)/gbplanner_config.yaml"/>
  <arg name="pci_file" default="$(find gbplanner)/config/$(arg robot_name)/planner_control_interface_sim_config.yaml"/>
  <arg name="voxblox_config_file" default="$(find gbplanner)/config/$(arg robot_name)/voxblox_sim_config.yaml"/>
  <arg name="map_config_file" default="$(arg voxblox_config_file)"/>
 
  <node pkg="tf" type="static_transform_publisher" name="tf_53" args="0 0 0 0 0 0 world navigation 100" />

  <arg name="odometry_topic" default="/state_estimation"/>  <arg name="pointcloud_topic" default="/velodyne_points"/>  <node pkg="gbplanner" type="gbplanner_node" name="gbplanner_node" output="screen">
    <remap from="odometry" to="$(arg odometry_topic)" />
    <remap from="/pointcloud" to="$(arg pointcloud_topic)" />
    <rosparam command="load" file="$(arg gbplanner_config_file)" />
    <rosparam command="load" file="$(arg map_config_file)" />
  </node>

  <node pkg="pci_general" type="pci_general_ros_node" name="pci_general_ros_node" output="screen">
    <remap from="command/trajectory" to="/trajectory" />  <remap from="planner_server" to="gbplanner" />
    <remap from="planner_homing_server" to="gbplanner/homing" />
    <remap from="odometry" to="$(arg odometry_topic)"/>
    <rosparam command="load" file="$(arg pci_file)" />
  </node>

  <node pkg="rviz" type="rviz" name="gbplanner_ui" output="screen" args="-d $(find gbplanner)/config/rviz/smb.rviz"/>  
  </launch>`

And when I run this file, the following error pops up: `process[tf_53-1]: started with pid [6151]
process[gbplanner_node-2]: started with pid [6152]
process[pci_general_ros_node-3]: started with pid [6157]
process[gbplanner_ui-4]: started with pid [6159]
[ INFO] [1738896894.393355941]: rviz version 1.14.25
[ INFO] [1738896894.393397068]: compiled against Qt version 5.12.8
[ INFO] [1738896894.393407197]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1738896894.401445710]: Forcing OpenGl version 0.
/gbplanner_node/SensorParams/VLP16 0, 0, 0
[ERROR] [1738896894.700832603, 8.960000000]: Client [/rvizGA] wants topic /trajectory to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [trajectory_msgs/MultiDOFJointTrajectory/ef145a45a5f47b77b7f5cdde4b16c942]. Dropping connection.
[ INFO] [1738896894.897918268, 9.151000000]: Stereo is NOT SUPPORTED
[ INFO] [1738896894.897970826, 9.151000000]: OpenGL device: NVIDIA GeForce RTX 4050 Laptop GPU/PCIe/SSE2
[ INFO] [1738896894.897983380, 9.151000000]: OpenGl version: 4.6 (GLSL 4.6).
[ERROR] [1738896894.980712102, 9.226000000]: Could not parse visual element for Link [base_origin]
[ INFO] [1738896906.058155970, 19.900000000]: Planning iteration 0
[ERROR] [1738896906.078341469, 19.920000000]: Exception thrown while processing service call: bool pcl::PCA<PointT>::initCompute() [with PointT = pcl::PointXYZI] in /usr/include/pcl-1.10/pcl/common/impl/pca.hpp @ 58 : [pcl::PCA::initCompute] number of points < 3
[ERROR] [1738896906.079731893, 19.921000000]: Service call failed: service [/gbplanner] responded with an error: bool pcl::PCA<PointT>::initCompute() [with PointT = pcl::PointXYZI] in /usr/include/pcl-1.10/pcl/common/impl/pca.hpp @ 58 : [pcl::PCA::initCompute] number of points < 3
[ERROR] [1738896906.079781486, 19.921000000]: Planner service failed
[ INFO] [1738896906.606308006, 20.431000000]: Planning iteration 0
`
I would be glad if you could help me resolve this issue. Thanks

ShayonMitra, 2025-02-06:
This is the launch file i am using. 
`<launch>

 <arg name="robot_name" default="smb"/>
  <arg name="gazebo_gui_en" default="true"/>
  <arg name="use_sim_time" default="true"/>
  <arg name="rviz_en" default="true" />
  <arg name="launch_prefix" default=""/> 
  <param name="use_sim_time" value="$(arg use_sim_time)"/>

  
  <arg name="gbplanner_config_file" default="$(find gbplanner)/config/$(arg robot_name)/gbplanner_config.yaml"/>
  <arg name="pci_file" default="$(find gbplanner)/config/$(arg robot_name)/planner_control_interface_sim_config.yaml"/>
  <arg name="voxblox_config_file" default="$(find gbplanner)/config/$(arg robot_name)/voxblox_sim_config.yaml"/>
  <arg name="map_config_file" default="$(arg voxblox_config_file)"/>
 
  <node pkg="tf" type="static_transform_publisher" name="tf_53" args="0 0 0 0 0 0 world navigation 100" />

  <arg name="odometry_topic" default="/state_estimation"/>  <arg name="pointcloud_topic" default="/velodyne_points"/>  <node pkg="gbplanner" type="gbplanner_node" name="gbplanner_node" output="screen">
    <remap from="odometry" to="$(arg odometry_topic)" />
    <remap from="/pointcloud" to="$(arg pointcloud_topic)" />
    <rosparam command="load" file="$(arg gbplanner_config_file)" />
    <rosparam command="load" file="$(arg map_config_file)" />
  </node>

  <node pkg="pci_general" type="pci_general_ros_node" name="pci_general_ros_node" output="screen">
    <remap from="command/trajectory" to="/trajectory" />  <remap from="planner_server" to="gbplanner" />
    <remap from="planner_homing_server" to="gbplanner/homing" />
    <remap from="odometry" to="$(arg odometry_topic)"/>
    <rosparam command="load" file="$(arg pci_file)" />
  </node>

  <node pkg="rviz" type="rviz" name="gbplanner_ui" output="screen" args="-d $(find gbplanner)/config/rviz/smb.rviz"/>  
  </launch>`

ShayonMitra, 2025-02-06:
This is the .launch file i am using. 

<launch>

 <arg name="robot_name" default="smb"/>
  <arg name="gazebo_gui_en" default="true"/>
  <arg name="use_sim_time" default="true"/>
  <arg name="rviz_en" default="true" />
  <arg name="launch_prefix" default=""/> 
  <param name="use_sim_time" value="$(arg use_sim_time)"/>

  
  <arg name="gbplanner_config_file" default="$(find gbplanner)/config/$(arg robot_name)/gbplanner_config.yaml"/>
  <arg name="pci_file" default="$(find gbplanner)/config/$(arg robot_name)/planner_control_interface_sim_config.yaml"/>
  <arg name="voxblox_config_file" default="$(find gbplanner)/config/$(arg robot_name)/voxblox_sim_config.yaml"/>
  <arg name="map_config_file" default="$(arg voxblox_config_file)"/>
 
  <node pkg="tf" type="static_transform_publisher" name="tf_53" args="0 0 0 0 0 0 world navigation 100" />

  <arg name="odometry_topic" default="/state_estimation"/>  <arg name="pointcloud_topic" default="/velodyne_points"/>  <node pkg="gbplanner" type="gbplanner_node" name="gbplanner_node" output="screen">
    <remap from="odometry" to="$(arg odometry_topic)" />
    <remap from="/pointcloud" to="$(arg pointcloud_topic)" />
    <rosparam command="load" file="$(arg gbplanner_config_file)" />
    <rosparam command="load" file="$(arg map_config_file)" />
  </node>

  <node pkg="pci_general" type="pci_general_ros_node" name="pci_general_ros_node" output="screen">
    <remap from="command/trajectory" to="/trajectory" />  <remap from="planner_server" to="gbplanner" />
    <remap from="planner_homing_server" to="gbplanner/homing" />
    <remap from="odometry" to="$(arg odometry_topic)"/>
    <rosparam command="load" file="$(arg pci_file)" />
  </node>

  <node pkg="rviz" type="rviz" name="gbplanner_ui" output="screen" args="-d $(find gbplanner)/config/rviz/smb.rviz"/>  
  </launch>

ShayonMitra, 2025-02-06:
This is smb_sim.launch file. I am using. 
[https://github.com/ShayonMitra/smb_sim_myversion.git](url)

MihirDharmadhikari, 2025-02-10:
Hi @ShayonMitra 

Seems like you are subscribing to the wrong message type on a topic. I am unable to access the files that you shared (the first one has no link and the second one gives a 404 error). If you can share the files again maybe I will be able to check what the issue is.

Best,
Mihir