ntnu-arl/gbplanner_ros #60: The path is abnormal in a large space.

romaster93, 2025-02-06:
I am testing gbplanner in a large space. 
When I start exploration, the drone keeps moving backward or sideways instead of proceeding forward. 
I want it to explore around walls or cover all areas without leaving any gaps. 
However, it does not move far from the starting point. I suspect that the issue might be the LiDAR's range being too wide, so I reduced the sensor's max range to 3 in the config file, but it did not make a difference. What should I modify?

*I have attached a test video. 

<iframe width="755" height="319" src="https://www.youtube.com/embed/-AxrWnZg5oA" title="gbplanner test v1 2025 02 06 18 32 22" frameborder="0" allow="accelerometer; autoplay; clipboard-write; encrypted-media; gyroscope; picture-in-picture; web-share" referrerpolicy="strict-origin-when-cross-origin" allowfullscreen></iframe>

MihirDharmadhikari, 2025-02-06:
Hi @romaster93 

It seems like the planner is trying to explore the space high up as it sees a lot of exploration gain by going higher. If you want it to explore that room and then exit (without going much higher), you should reduce the height of the global bounding box [here](https://github.com/ntnu-arl/gbplanner_ros/blob/7301b5350915cf3a4055072ad559842db37b4ca9/gbplanner/config/rmf_obelix/gbplanner_config.yaml#L40) (the third number).

Let me know if this works.

Best,
Mihir

romaster93, 2025-02-07:
Thanks! its work!

https://youtu.be/bGlJRBq-riY?si=60n0IAh-BYfxldGf

However, there are still areas that have not been visited. While point cloud data is collected from a distance, we want to ensure that as many areas as possible are visited. To achieve denser coverage, we modified the sensor's max range to 3 in the config file, but the issue persists. Are there any other parameters that need to be adjusted?

MihirDharmadhikari, 2025-02-10:
Hi @romaster93 

According to the map, it seems like the entire room is mapped. I don't see which unvisited parts you refer to. Can you elaborate on it?
If you want the robot to map all surfaces from 3m distance that will require some change.
The easiest option is to set the max mapping distance to 3m [here](https://github.com/ntnu-arl/gbplanner_ros/blob/7301b5350915cf3a4055072ad559842db37b4ca9/gbplanner/config/rmf_obelix/voxblox_sim_config.yaml#L6). But this will make the exploration very slow as the robot is now only seeing upto 3m.

Best,
Mihir

romaster93, 2025-02-12:
Hi @MihirDharmadhikari

romaster93, 2025-02-12:
Hi, @MihirDharmadhikari 

Thanks to reply!!

After modifying the parameters you provided to adjust the distance limitation for map collection, the exploration speed has decreased, but the number of visited locations has increased. This is exactly the result I wanted. Thank you!

I have an additional question.
After mapping is completed, when I input waypoints and start the plan to waypoint, the system generates an inefficient path even in open spaces without obstacles. Is there a parameter that can be adjusted to fix this issue?

https://youtu.be/lQtn2x7slLE?si=GGOQdeMOS2_EQcR-